pschatzmann/arduino-audio-driver #16: Audio kit keeps rebooting

deladriere, 2024-03-08:
I have the ESP32 Audio kit V2.2 and I am running this code:
```
#include <Arduino.h>

/**
 * @brief We use a predefied board (AudioKitEs8388V1) and output_device a sine
 * with the help of the AudioTools I2SCodecStream
 * @author phil schatzmann
 */

#include "AudioTools.h" // install https://github.com/pschatzmann/arduino-audio-tools
#include "AudioLibs/I2SCodecStream.h"

AudioInfo info(44100, 2, 16);
SineWaveGenerator<int16_t> sineWave(32000);
GeneratedSoundStream<int16_t> sound(sineWave);
I2SCodecStream out(AudioKitEs8388V1);
StreamCopy copier(out, sound);

void setup()
{
  // Setup logging
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  LOGLEVEL_AUDIODRIVER = AudioDriverWarning;

  // start I2S & codec with i2c and i2s configured above
  Serial.println("starting I2S...");
  auto config = out.defaultConfig();
  config.copyFrom(info);
  out.begin(config);

  // Setup sine wave
  sineWave.begin(info, N_B4);
}

// Arduino loop - copy sound to out
void loop() { copier.copy(); }
```
on PlatformIO
with this platformio.ini

```

[platformio]
description = Audio Example
default_envs = esp32dev

[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
lib_deps =  
    https://github.com/pschatzmann/arduino-audio-tools.git
    https://github.com/pschatzmann/arduino-audio-driver.git
lib_ldf_mode = deep+
build_flags =  -DCORE_DEBUG_LEVEL=5 -DAUDIOKIT_BOARD=5 
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
````
Compile and upload work fine, but it keeps rebooting with this message :

```

ELF file SHA256: 11734f0e505573e4

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    23][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
starting I2S...
Warning: Pin '13' not set up because of conflict
[    35][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=33 scl=32 freq=100000
Error:   ->p_wire->endTransmission: 2
Error:   ES8388 not found on I2C address 10, check wiring
Error:   AudioDriver::begin::init failed
Error:   ->p_wire->endTransmission: 2
Error:   ->p_wire->endTransmission: 2
Error:   ->p_wire->endTransmission: 2
Error:   ->p_wire->endTransmission: 2
Error:   ->p_wire->endTransmission: 2
Error:   ->p_wire->endTransmission: 2
[E] I2SCodecStream.h : 73 - virtual bool audio_tools::I2SCodecStream::begin()
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400e1cda  PS      : 0x00060a30  A0      : 0x800d3f4b  A1      : 0x3ffb2120  
A2      : 0x00000000  A3      : 0x3ffba8a0  A4      : 0x00000400  A5      : 0x3ffb2160  
A6      : 0xffffffff  A7      : 0x3ffc2f24  A8      : 0x00000000  A9      : 0x3ffb2120  
A10     : 0x3ffb2164  A11     : 0x00000010  A12     : 0x40ac101f  A13     : 0x37354443  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000018  LBEG    : 0x400029ac  LEND    : 0x400029cb  LCOUNT  : 0x00000000  

Backtrace: 0x400e1cd7:0x3ffb2120 0x400d3f48:0x3ffb2160 0x400d3ff9:0x3ffb21d0 0x400d4039:0x3ffb21f0 0x400d5cd7:0x3ffb2210 0x400d5e90:0x3ffb2230 0x400d5fa6:0x3ffb2270 0x400de4a9:0x3ffb2290

ELF file SHA256: 11734f0e505573e4

Rebooting...

```
What am I doing wrong?
Thanks!

pschatzmann, 2024-03-08:
AudioKitEs8388V1 must be wrong for your board!
Did you try AudioKitEs8388V1 ?

deladriere, 2024-03-08:
aha thanks !
it works, but the serial still reports warning :
```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    12][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
starting I2S...
Warning: Pin '13' not set up because of conflict
Warning: Pin '23' not set up because of conflict
Warning: Pin '18' not set up because of conflict
[    57][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=18 scl=23 freq=100000
```

pschatzmann, 2024-03-08:
That's a feature of your board:  The same pins are used for different things, which leads to conflicts...

deladriere, 2024-03-10:
ah ok thanks !

closed 2024-03-10 by deladriere

Frank-Bemelman, 2025-03-08:
Apologies for re-opening, but I have more or less the same problem, except the suggestions don't work. My board is ESP32 Audio Kit V2.2 A237

Same code as above, including same platformio.ini 
My #include is slightly different:
#include "AudioTools/AudioLibs/I2SCodecStream.h"
... because there is where Platformio has installed this lib.

Compiles without problems. But my board keeps rebooting:

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
[    24][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    41][V][esp32-hal-uart.c:330] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[    51][V][esp32-hal-uart.c:416] uartBegin(): UART0 not installed. Starting installation
[    62][V][esp32-hal-uart.c:467] uartBegin(): UART0 initialization done.
starting I2S...
Info:    AudioDriver::begin
Info:    setupI2CAddress: 0x10
Info:    ==> Updating address: 0x10
Warning: Pin '13' not set up because of conflict
Warning: Pin '23' not set up because of conflict
Warning: Pin '18' not set up because of conflict
Info:    setting up SPI miso:2,mosi:15, clk:14, cs:13
Info:    setting up I2C scl: 23, sda: 18
[    99][E][Wire.cpp:297] begin(): Bus already started in Slave Mode.

assert failed: heap_caps_free heap_caps.c:381 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x40083c1d:0x3ffb1df0 0x4008a4a5:0x3ffb1e10 0x4008f629:0x3ffb1e30 0x40084062:0x3ffb1f60 0x4008f659:0x3ffb1f80 0x400dacf5:0x3ffb1fa0 0x400dae82:0x3ffb1fc0 0x400df3b9:0x3ffb1fe0 0x400d69e3:0x3ffb2000 0x400d6ce0:0x3ffb2020 0x400d7ab6:0x3ffb2070 0x400d7b81:0x3ffb20f0 0x400d7cc9:0x3ffb2170 0x400d7dcb:0x3ffb21a0 0x400e32ce:0x3ffb2290
  #0  0x40083c1d in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x4008a4a5 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x4008f629 in __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c:85
  #3  0x40084062 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:381
      (inlined by) heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:366
  #4  0x4008f659 in free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/heap.c:39
  #5  0x400dacf5 in TwoWire::freeWireBuffer() at C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp:171
  #6  0x400dae82 in TwoWire::begin(int, int, unsigned int) at C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/libraries/Wire/src/Wire.cpp:316 (discriminator 1)
  #7  0x400df3b9 in i2c_bus_create at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-driver/src/Platforms/API_I2C_Arduino.cpp:22
  #8  0x400d69e3 in audio_driver::PinsI2C::begin() at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-driver/src/DriverPins.h:198
      (inlined by) audio_driver::PinsI2C::begin() at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-driver/src/DriverPins.h:195
      (inlined by) audio_driver::DriverPins::begin() at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-driver/src/DriverPins.h:383
  #9  0x400d6ce0 in audio_driver::AudioDriver::begin(audio_driver::CodecConfig, audio_driver::DriverPins&) at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-driver/src/Driver.h:205
  #10 0x400d7ab6 in audio_driver::AudioBoard::begin() at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-driver/src/AudioBoard.h:32
      (inlined by) audio_driver::AudioBoard::begin(audio_driver::CodecConfig) at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-driver/src/AudioBoard.h:44       
      (inlined by) audio_tools::I2SCodecStream::beginCodec(audio_tools::I2SCodecConfig) at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioLibs/I2SCodecStream.h:312
  #11 0x400d7b81 in audio_tools::I2SCodecStream::begin1() at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioLibs/I2SCodecStream.h:219 (discriminator 2)
  #12 0x400d7cc9 in audio_tools::I2SCodecStream::begin(audio_tools::I2SCodecConfig) at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/.pio/libdeps/esp32dev/audio-tools/src/AudioTools/AudioLibs/I2SCodecStream.h:82 (discriminator 2)
  #13 0x400d7dcb in setup() at C:\Users\Frank\Documents\platformio\projects\Frank Internet Radio/src/main.cpp:26
  #14 0x400e32ce in loopTask(void*) at C:/Users/Frank/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:42

ELF file SHA256: 5895bb46eb790dde

Rebooting...
ets Jul 29 2019 12:21:46
(and so on)

No idea how to attack this....
Any ideas?

Thanks in advance,
Frank

pschatzmann, 2025-03-09:
This is completely different! 
Please open a separate issue and provide the necessary information...