pschatzmann/arduino-audio-driver #18: Encoding failed: Handle passed to function call was invalid

2e0wkw, 2024-03-19:
Hi, I am using the codec with Arduino Audio Tools on an Audio Kit board.

[E] AACEncoderFDK.h : 234 - Encoding failed: Handle passed to function call was invalid

I am using the streams-audiokit-webserver_aac.ino example.
I am also using the new Auduino-Audio-Driver.
I am also using the  auduino-fdk-acc codec

The code builds and uploads but i get this runtime error in the serial monitor when i connect to the web server via Chrome.

Thanks

Eric

2e0wkw, 2024-03-19:
Hi i am using the Audio Kit with ESP32 A1S. I have switched on logging and see that it related to a memory issue. I have read you blogs about Audio Kit and Arduino but i am still missing sone of the basic setup for using the board with arduino.

Thanks

pschatzmann, 2024-03-19:
Could you please answer my questions: The core version you can see in the baord manager when you select ESP32 and the psram settings you can find in the Tools menu in Arduino

2e0wkw, 2024-03-19:
The board i select is the ESP32 Dev Board (i could not find one for A1S) and PSRAM is disabled at the moment

pschatzmann, 2024-03-19:
I need the version number! 
Without this I will not be able to reproduce your result..

2e0wkw, 2024-03-19:
ah sorry my mistake 2.0.13

pschatzmann, 2024-03-19:
I could not reproduce your error with [this example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-audiokit/streams-audiokit-webserver_aac/streams-audiokit-webserver_aac.ino)

closed 2024-03-19 by pschatzmann

2e0wkw, 2024-03-19:
Thanks fir trying Phil. To be honest i have just got it working by looking into all the things you asked me about. I think PSRAM was the main thing and updating the arduino/esp core to 2.0.14