pschatzmann/arduino-audio-driver #27: How can I use ES8311 ?

rushairer, 2024-07-30:
I copied the code from example anc change it like that:

```cpp

#include <Arduino.h>
#include "AudioTools.h"
#include "AudioLibs/I2SCodecStream.h"
#include <Wire.h>

// I2C
#define SDAPIN 3        // I2C Data,  Adafruit ESP32 S3 3, Sparkfun Thing Plus C 23
#define SCLPIN 4        // I2C Clock, Adafruit ESP32 S3 4, Sparkfun Thing Plus C 22
#define I2CSPEED 100000 // Clock Rate
#define ES8311ADDR 0x19 // Address of ES8311 I2C port

// I2S, your configuration for the ES8311 board
#define MCLKPIN 14 // Master Clock
#define BCLKPIN 13 // Bit Clock
#define WSPIN 8    // Word select
#define DOPIN 7    // This is connected to DI on ES8311 (MISO)
#define DIPIN 9    // This is connected to DO on ES8311 (MOSI)

AudioDriverES8311Class es8311(ES8311ADDR);
AudioInfo audio_info(44200, 2, 16);                    // sampling rate, # channels, bit depth
SineWaveGenerator<int16_t> sine_wave(32000);           // amplitude
GeneratedSoundStream<int16_t> sound_stream(sine_wave); // sound generator
DriverPins my_pins;                                    // board pins
AudioBoard audio_board(es8311, my_pins);               // audio board
I2SCodecStream i2s_out_stream(audio_board);            // i2s coded
StreamCopy copier(i2s_out_stream, sound_stream);       // stream copy sound generator to i2s codec
TwoWire myWire = TwoWire(0);                           // universal I2C interface

void setup()
{
    // Setup logging
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Debug);
    LOGLEVEL_AUDIODRIVER = AudioDriverDebug;
    delay(2000);

    Serial.println("Setup starting...");

    Serial.println("I2C pin ...");
    my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES8311ADDR, I2CSPEED, myWire, true);
    Serial.println("I2S pin ...");
    my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, DOPIN, DIPIN);

    Serial.println("Pins begin ...");
    my_pins.begin();

    Serial.println("Board begin ...");
    audio_board.begin();

    Serial.println("I2S begin ...");
    auto i2s_config = i2s_out_stream.defaultConfig();
    i2s_config.copyFrom(audio_info);

    i2s_out_stream.begin(i2s_config); // this should apply I2C and I2S configuration

    // Setup sine wave
    Serial.println("Sine wave begin...");
    sine_wave.begin(audio_info, N_B4); // 493.88 Hz

    Serial.println("Setup completed ...");
}

// Arduino loop - copy sound to out
void loop()
{
    copier.copy();
}

```

I got the error:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4205b0e2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
Setup starting...
I2C pin ...
I2S pin ...
Pins begin ...
Debug:   DriverPins::begin
Debug:   DriverPins::begin::setupPinMode
Debug:   DriverPins::begin::SPI
Debug:   DriverPins::begin::I2C
Debug:   DriverPins::begin::I2C port:25
Debug:   PinsI2C::begin for 7
Info:    setting up I2C scl: 4, sda: 3
[  2175][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=3 scl=4 freq=100000
Info:    Setting i2c clock: 100000
Board begin ...
Debug:   AudioBoard::pins::begin
Debug:   DriverPins::begin
Debug:   DriverPins::begin::setupPinMode
Debug:   DriverPins::begin::SPI
Debug:   DriverPins::begin::I2C
Debug:   DriverPins::begin::I2C port:25
Debug:   PinsI2C::begin for 7
Info:    setting up I2C scl: 4, sda: 3
[  2176][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
Info:    Setting i2c clock: 100000
Debug:   AudioBoard::pins::begin::returned:true
Debug:   AudioBoard::driver::begin
Debug:   AudioDriver::begin:pins
Debug:   AudioDriver::begin:setSPI
Debug:   AudioDriver::begin:setConfig
Error:   AudioDriver::begin::init failed
Debug:   AudioDriver::begin:setPAPower
Debug:   AudioDriver::begin:completed
Debug:   SET: volume:178

assert failed: audio_driver::error_t i2c_bus_write_bytes(audio_driver::i2c_bus_handle_t, int, uint8_t*, int, uint8_t*, int) I2C.cpp:14 (p_wire != nullptr)

Backtrace: 0x40377cbe:0x3fcebc80 0x4037afc5:0x3fcebca0 0x40380af1:0x3fcebcc0 0x4200d641:0x3fcebdf0 0x4200adb2:0x3fcebe10 0x4200b2c7:0x3fcebe40 0x42002f9b:0x3fcebe60 0x42005bb9:0x3fcebe80 0x42005f1b:0x3fcebec0 0x42007c28:0x3fcebf10 0x420101fe:0x3fcec020

```

It means p_wire is null?
so, i add 

```cpp
CodecConfig cfg;
audio_board.setConfig(cfg);
```
before ` audio_board.begin();`

The I got the error log: 

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42019f3e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a68
entry 0x403c98d4
Setup starting...
I2C pin ...
I2S pin ...
Pins begin ...
Debug:   DriverPins::begin
Debug:   DriverPins::begin::setupPinMode
Debug:   DriverPins::begin::SPI
Debug:   DriverPins::begin::I2C
Debug:   DriverPins::begin::I2C port:25
Debug:   PinsI2C::begin for 7
Info:    setting up I2C scl: 4, sda: 3
[  2174][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=3 scl=4 freq=100000
Info:    Setting i2c clock: 100000
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x42005333  PS      : 0x00060630  A0      : 0x82005620  A1      : 0x3fcebe20  
A2      : 0x00000000  A3      : 0x0000000b  A4      : 0x00000000  A5      : 0x00000100  
A6      : 0x3fc96da0  A7      : 0x00000000  A8      : 0x8200da99  A9      : 0x3fcebe40  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x3fcec350  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000040  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0x00000000  

Backtrace: 0x42005330:0x3fcebe20 0x4200561d:0x3fcebe60 0x42005aa2:0x3fcebe80 0x42005f75:0x3fcebec0 0x42007c51:0x3fcebf00 0x4201022a:0x3fcec020
```

Is there anything wrong?

Thanks!
[PCB_ES8311_2024-07-30.pdf](https://github.com/user-attachments/files/16421733/PCB_ES8311_2024-07-30.pdf)
[image]

pschatzmann, 2024-07-30:
- Error:   AudioDriver::begin::init failed because you forgot to define the masterclock pin.
- Dont call  my_pins.begin(); and  audio_board.begin(): As you can see these functions are executed twice now
- Just use one of the predefined Wire objects [Wire or Wire1
](https://github.com/espressif/arduino-esp32/blob/master/libraries/Wire/src/Wire.h)

pschatzmann, 2024-07-30:
I committed a correction that prints out a message and if nothing is defined assumes to use FROM_MCLK_PIN (=0)

rushairer, 2024-07-30:
Thank you very much for your help

What I don't quite understand is:
`my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, DOPIN, DIPIN);`
Isn't this command defining the masterclock source?

What is the correct way? I didn't find it in the example

pschatzmann, 2024-07-30:
It seems that  es8311_codec_init(codec_config_t *codec_cfg, i2c_bus_handle_t handle, int8_t mclk_src, int i2c_address); is asking for a mclk_src with the following values

/*
 * to define the clock soure of MCLK
 */
#define FROM_MCLK_PIN       0
#define FROM_SCLK_PIN       1

It seems that I used the pin definitions to check and define this. So you can use
pins.addPin(MCLK_SOURCE, 0);

rushairer, 2024-07-30:
```cpp
#include <Arduino.h>
#include "AudioTools.h"
#include "AudioLibs/I2SCodecStream.h"
#include <Wire.h>

// I2C
#define SDAPIN 3        // I2C Data,  Adafruit ESP32 S3 3, Sparkfun Thing Plus C 23
#define SCLPIN 4        // I2C Clock, Adafruit ESP32 S3 4, Sparkfun Thing Plus C 22
#define I2CSPEED 100000 // Clock Rate
#define ES8311ADDR 0x19 // Address of ES8311 I2C port

// I2S, your configuration for the ES8311 board
#define MCLKPIN 14 // Master Clock
#define BCLKPIN 13 // Bit Clock
#define WSPIN 8    // Word select
#define DOPIN 7    // This is connected to DI on ES8311 (MISO)
#define DIPIN 9    // This is connected to DO on ES8311 (MOSI)

// AudioDriverES8311Class es8311(ES8311ADDR);
AudioInfo audio_info(44200, 2, 16);                    // sampling rate, # channels, bit depth
SineWaveGenerator<int16_t> sine_wave(32000);           // amplitude
GeneratedSoundStream<int16_t> sound_stream(sine_wave); // sound generator
DriverPins my_pins;                                    // board pins
AudioBoard audio_board(AudioDriverES8311, my_pins);    // audio board
I2SCodecStream i2s_out_stream(audio_board);            // i2s coded
StreamCopy copier(i2s_out_stream, sound_stream);       // stream copy sound generator to i2s codec

void setup()
{
    // Setup logging
    Serial.begin(115200);
    Wire.begin(SDAPIN, SCLPIN);

    AudioLogger::instance().begin(Serial, AudioLogger::Warning);
    LOGLEVEL_AUDIODRIVER = AudioDriverWarning;
    delay(3000);

    Serial.println("Setup starting...");

    Serial.println("I2C pin ...");
    my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES8311ADDR, I2CSPEED, Wire, true);

    Serial.println("I2S pin ...");
    my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, DOPIN, DIPIN);
    my_pins.addPin(PinFunction::MCLK_SOURCE, MCLKPIN, PinLogic::Output);

    Serial.println("Pins begin ...");
    // my_pins.begin();

    Serial.println("Board begin ...");
    audio_board.begin();

    Serial.println("I2S begin ...");
    auto i2s_config = i2s_out_stream.defaultConfig();
    i2s_config.copyFrom(audio_info);
    i2s_out_stream.begin(i2s_config); // this should apply I2C and I2S configuration

    // Setup sine wave
    Serial.println("Sine wave begin...");
    sine_wave.begin(audio_info, N_B4); // 493.88 Hz

    Serial.println("Setup completed ...");
}

// Arduino loop - copy sound to out
void loop()
{
    copier.copy();
}

```

I added `my_pins.addPin(PinFunction::MCLK_SOURCE, MCLKPIN, PinLogic::Output);`
using `Wire.begin(SDAPIN, SCLPIN);` and `my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES8311ADDR, I2CSPEED, Wire, true);`

Got 

```
I2C pin ...
I2S pin ...
Pins begin ...
Board begin ...
[  3174][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
I2S begin ...
[  3198][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
Error:   ->p_wire->endTransmission: 2
[  3224][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
Error:   ->p_wire->requestFrom 1->0
Error:   ->p_wire->endTransmission: 2
[  3224][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
Error:   ->p_wire->requestFrom 1->0
Error:   ->p_wire->endTransmission: 2
Sine wave begin...
Setup completed ...
```

If i use Wire1 , add `build_flags = -D WIRE_INTERFACES_COUNT=2`

```cpp
    Wire1.begin(SDAPIN, SCLPIN);
....
    my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES8311ADDR, I2CSPEED, Wire1, true);

```
Got 

```
Setup starting...
I2C pin ...
I2S pin ...
Pins begin ...
Board begin ...
[  3174][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
I2S begin ...
[  3198][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
Error:   ->p_wire->endTransmission: 2
[  3224][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
Error:   ->p_wire->requestFrom 1->0
Error:   ->p_wire->endTransmission: 2
[  3224][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
Error:   ->p_wire->requestFrom 1->0
Error:   ->p_wire->endTransmission: 2
Sine wave begin...
Setup completed ...
```

Am I using Wire incorrectly?

pschatzmann, 2024-07-30:
You don't need to call Wire1.begin(SDAPIN, SCLPIN); Therefore you are getting some warnings subsequently.

If you get errors there must be something wrong with your wire definitions...

pschatzmann, 2024-07-30:
Use the WireScan sketch to check the I2C  pins and to confirm the port

pschatzmann, 2024-07-30:
ps. the PinFunction::MCLK_SOURCE is still wrong: the valid values are 0 or 1

rushairer, 2024-07-30:
Thanks again. I am an amateur electronics enthusiast and I don't know much about the basics. I will try to modify it later. I mixed up MCLK_SOURCE and MCLK_PIN

rushairer, 2024-07-30:
I  use 
```cpp

#include <Arduino.h>
#include "Wire.h"
#include "SPI.h"

void setup()
{
    Serial.begin(115200);
    Wire.begin(3, 4);
}

void loop()
{
    byte error, address;
    int nDevices = 0;

    delay(5000);

    Serial.println("Scanning for I2C devices ...");
    for (address = 0x01; address < 0x7f; address++)
    {
        Wire.beginTransmission(address);
        error = Wire.endTransmission();
        if (error == 0)
        {
            Serial.printf("I2C device found at address 0x%02X\n", address);
            nDevices++;
        }
        else if (error != 2)
        {
            Serial.printf("Error %d at address 0x%02X\n", error, address);
        }
    }
    if (nDevices == 0)
    {
        Serial.println("No I2C devices found");
    }
}

```

Got `0x19`

```cpp

#include <Arduino.h>
#include "AudioTools.h"
#include "AudioLibs/I2SCodecStream.h"
#include <Wire.h>

// I2C
#define SDAPIN 3        // I2C Data,  Adafruit ESP32 S3 3, Sparkfun Thing Plus C 23
#define SCLPIN 4        // I2C Clock, Adafruit ESP32 S3 4, Sparkfun Thing Plus C 22
#define I2CSPEED 100000 // Clock Rate
#define ES8311ADDR 0x19 // Address of ES8311 I2C port

// I2S, your configuration for the ES8311 board
#define MCLKPIN 14 // Master Clock
#define BCLKPIN 13 // Bit Clock
#define WSPIN 8    // Word select
#define DOPIN 7    // This is connected to DI on ES8311 (MISO)
#define DIPIN 9    // This is connected to DO on ES8311 (MOSI)

// AudioDriverES8311Class es8311(ES8311ADDR);
AudioInfo audio_info(44200, 2, 16);                    // sampling rate, # channels, bit depth
SineWaveGenerator<int16_t> sine_wave(32000);           // amplitude
GeneratedSoundStream<int16_t> sound_stream(sine_wave); // sound generator
DriverPins my_pins;                                    // board pins
AudioBoard audio_board(AudioDriverES8311, my_pins);    // audio board
I2SCodecStream i2s_out_stream(audio_board);            // i2s coded
StreamCopy copier(i2s_out_stream, sound_stream);       // stream copy sound generator to i2s codec

void setup()
{
    // Setup logging
    Serial.begin(115200);

    AudioLogger::instance().begin(Serial, AudioLogger::Warning);
    LOGLEVEL_AUDIODRIVER = AudioDriverWarning;
    delay(3000);

    Serial.println("Setup starting...");

    Serial.println("I2C pin ...");
    my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES8311ADDR, I2CSPEED, Wire, true);

    Serial.println("I2S pin ...");
    my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, DOPIN, DIPIN);
    my_pins.addPin(PinFunction::MCLK_SOURCE, 0, PinLogic::Output);

    Serial.println("Board begin ...");
    audio_board.begin();

    Serial.println("I2S begin ...");
    auto i2s_config = i2s_out_stream.defaultConfig();
    i2s_config.copyFrom(audio_info);
    i2s_out_stream.begin(i2s_config); // this should apply I2C and I2S configuration

    // Setup sine wave
    Serial.println("Sine wave begin...");
    sine_wave.begin(audio_info, N_B4); // 493.88 Hz

    Serial.println("Setup completed ...");
}

// Arduino loop - copy sound to out
void loop()
{
    copier.copy();
}

```

Got

```
Setup starting...
I2C pin ...
I2S pin ...
Board begin ...
[  3174][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=3 scl=4 freq=100000
I2S begin ...
[  3198][W][Wire.cpp:301] begin(): Bus already started in Master Mode.
Error:   ->p_wire->endTransmission: 2
[  3224][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
Error:   ->p_wire->requestFrom 1->0
Error:   ->p_wire->endTransmission: 2
[  3225][E][Wire.cpp:513] requestFrom(): i2cRead returned Error -1
Error:   ->p_wire->requestFrom 1->0
Error:   ->p_wire->endTransmission: 2
Sine wave begin...
Setup completed ...
```

rushairer, 2024-07-31:
Change Log to Debug mode.

```
[D] I2SESP32.h : 96 - size_t audio_tools::I2SDriverESP32::writeBytes(const void*, size_t)
[D] I2SESP32.h : 104 - i2s_write 1024 -> 1024 bytes
[D] StreamCopy.h : 390 - write: 1024 -> 1024
[I] StreamCopy.h : 158 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
[D] StreamCopy.h : 98 - copy 1024 bytes 
[D] StreamCopy.h : 192 - available: 2048
[D] AudioStreams.h : 446 - GeneratedSoundStream::readBytes: 1024
[D] SoundGenerator.h : 62 - readBytes: 1024
[D] StreamCopy.h : 384 - write: 1024
[D] I2SCodecStream.h : 122 - I2SStream::write: 1024
[D] I2SStream.h : 114 - I2SStream::write: 1024
[D] I2SESP32.h : 96 - size_t audio_tools::I2SDriverESP32::writeBytes(const void*, size_t)
[D] I2SESP32.h : 104 - i2s_write 1024 -> 1024 bytes
[D] StreamCopy.h : 390 - write: 1024 -> 1024
[I] StreamCopy.h : 158 - StreamCopy::copy  1024 -> 1024 -> 1024 bytes - in 1 hops
[D] StreamCopy.h : 98 - copy 1024 bytes 
[D] StreamCopy.h : 192 - available: 2048
[D] AudioStreams.h : 446 - GeneratedSoundStream::readBytes: 1024
[D] SoundGenerator.h : 62 - readBytes: 1024
[D] StreamCopy.h : 384 - write: 1024
[D] I2SCodecStream.h : 122 - I2SStream::write: 1024
[D] I2SStream.h : 114 - I2SStream::write: 1024
```

Does the appearance of these logs mean that the program is running normally? But there is no sound, do I need to re-check the speaker circuit?

pschatzmann, 2024-07-31:
As I was writing above, don't continue if i2s_out_stream.begin(i2s_config); returns false

pschatzmann, 2024-08-19:
Closed due to inactivity

closed 2024-08-19 by pschatzmann