pschatzmann/arduino-audio-driver #32: ES8374 Codec Initialization Error on ESP32-S3-Mini 1: AudioDriver::begin::init failed

hakkican, 2024-08-28:
Hello,

I am using the ES8374 codec with an ESP32-S3-Mini 1. I made the connections based on the ESP32-A1S model. I did an I2C scan, and I can see that the ES8374 is accessible. I have modified the audiotools-custom-max.ino file for the ES8374. When I compile the code, I get the following error: "Warning: The ES8374 codec has already been initialized! Error: AudioDriver::begin::init failed." I reviewed the code step by step, and in the es8374.c file, the function es8374_codec_initialized() returns true at line 688, causing the process to exit.

Could you please help me with this issue?

Thank you.

Arduino IDE 2.3.2 
ESP32 by espressif 3.0.4 
arduino-audio-tools and arduino-audio-driver are up to date

ino file
```c
/**
 * @brief We define a custom board with the i2c and i2s pins and output_device a sine
 * with the help of the AudioTools I2SCodecStream
 * @author phil schatzmann
 */

#include "AudioTools.h" // install https://github.com/pschatzmann/arduino-audio-tools
#include "AudioLibs/I2SCodecStream.h"

// I2C
#define SDAPIN               3 // I2C Data,  Adafruit ESP32 S3 3, Sparkfun Thing Plus C 23
#define SCLPIN               4 // I2C Clock, Adafruit ESP32 S3 4, Sparkfun Thing Plus C 22
#define I2CSPEED        100000 // Clock Rate
#define ES8374ADDR        0x10 // Address of ES8374 I2C port

// I2S, your configuration for the ES8374 board
#define MCLKPIN             14 // Master Clock
#define BCLKPIN             36 // Bit Clock
#define WSPIN                8 // Word select
#define DOPIN               37 // This is connected to DI on ES8388 (MISO)
#define DIPIN               35 // This is connected to DO on ES8388 (MOSI)

AudioInfo                     audio_info(44200, 1, 16);                // sampling rate, # channels, bit depth
SineWaveGenerator<int16_t>    sine_wave(32000);                        // amplitude
GeneratedSoundStream<int16_t> sound_stream(sine_wave);                 // sound generator
DriverPins                    my_pins;                                 // board pins
AudioBoard                    audio_board(AudioDriverES8374, my_pins); // audio board
I2SCodecStream                i2s_out_stream(audio_board);             // i2s coded
StreamCopy                    copier(i2s_out_stream, sound_stream);    // stream copy sound generator to i2s codec
TwoWire                       myWire = TwoWire(0);                     // universal I2C interface

void setup() {
  // Setup logging
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);//Debug);
  LOGLEVEL_AUDIODRIVER = AudioDriverDebug;//AudioDriverWarning;
  delay(2000);

  Serial.println("Setup starting...");

  Serial.println("I2C pin ...");
  my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES8374ADDR, I2CSPEED, myWire);
  Serial.println("I2S pin ...");
  my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, DOPIN, DIPIN);

  Serial.println("Pins begin ..."); 
  my_pins.begin();

  Serial.println("Board begin ..."); 
  audio_board.begin();

  Serial.println("I2S begin ..."); 
  auto i2s_config = i2s_out_stream.defaultConfig();
  //i2s_config.output_device=DAC_OUTPUT_LINE1; 
  //i2s_config.sd_active = false; 
  i2s_config.copyFrom(audio_info);  
  i2s_out_stream.begin(i2s_config); // this should apply I2C and I2S configuration

  // Setup sine wave
  Serial.println("Sine wave begin...");
  sine_wave.begin(audio_info, N_B4); // 493.88 Hz

  Serial.println("Setup completed ...");
}

// Arduino loop - copy sound to out
void loop() { copier.copy(); }
```

log
```
...
I2S begin ...
Debug:   AudioBoard::pins::begin
Debug:   DriverPins::begin
Debug:   DriverPins::begin::setupPinMode
Debug:   DriverPins::begin::SPI
Debug:   DriverPins::begin::I2C
Debug:   DriverPins::begin::I2C port:16
Debug:   PinsI2C::begin for 7
Info:    setting up I2C scl: 4, sda: 3
Info:    Setting i2c clock: 100000
Debug:   AudioBoard::pins::begin::returned:true
Debug:   AudioBoard::driver::begin
Debug:   AudioDriver::begin:pins
Debug:   AudioDriver::begin:setSPI
Debug:   AudioDriver::begin:setConfig
Debug:   mode->CODEC_MODE_DECODE
Warning: The es8374 codec has already been initialized!
Error:   AudioDriver::begin::init failed
Debug:   AudioDriver::begin:setPAPower
Debug:   AudioDriver::begin:completed
Debug:   volume 70
Debug:   i2c_bus_write_bytes: addr=0x10 reglen=1 datalen=1 - reg=0x38 value=0x34
Debug:   AudioBoard::driver::begin::returned:false
Debug:   volume 70
Debug:   i2c_bus_write_bytes: addr=0x10 reglen=1 datalen=1 - reg=0x38 value=0x34
Debug:   AudioBoard::volume::set
[E] I2SCodecStream.h : 209 - bool audio_tools::I2SCodecStream::begin1()
Sine wave begin...
Setup completed ...
[E] StreamCopy.h : 401 - write  to target has failed after 22 retries! (1024 bytes)
[E] StreamCopy.h : 401 - write  to target has failed after 22 retries! (1024 bytes)
[E] StreamCopy.h : 401 - write  to target has failed after 22 retries! (1024 bytes)
```

pschatzmann, 2024-08-28:
Don't do this:

  Serial.println("Pins begin ..."); 
  my_pins.begin();

  Serial.println("Board begin ..."); 
  audio_board.begin();

This is done as part of   audio_board.begin();

closed 2024-08-28 by pschatzmann

hakkican, 2024-08-28:
Problem solved. Thanks