pschatzmann/arduino-audio-driver #41: How to use ES7210 to sample 4 channel audio?

xiqwertyui, 2024-12-12:
I need to use ES7210 to sample 4 adc microphones then write the data into sd card.
First, I try to start the codec chips but fail.

`#include "AudioTools.h"  // install https://github.com/pschatzmann/arduino-audio-tools
#include "AudioTools/AudioLibs/I2SCodecStream.h"
#include "HAL.h"

// I2C configuration
#define SDAPIN CONFIG_SDA  // I2C Data pin
#define SCLPIN CONFIG_SCL  // I2C Clock pin
#define I2CSPEED 100000    // I2C Clock Rate
#define ES7210ADDR 0x40    // Address of ES7210 I2C port

// I2S configuration
#define MCLKPIN CONFIG_ES7210_MCLK   // Master Clock
#define BCLKPIN CONFIG_ES7210_SCLK   // Bit Clock
#define WSPIN CONFIG_ES7210_LRCK     // Word Select (LRCK)
#define SDOUT1 CONFIG_ES7210_SDOUT1  // Data Out 1
#define SDOUT2 CONFIG_ES7210_SDOUT2  // Data Out 2

AudioInfo audio_info(44200, 4, 16);
DriverPins my_pins;
AudioBoard audio_board(AudioDriverES7210, my_pins);
I2SCodecStream i2s_in_stream(audio_board);

extern int LOGLEVEL_AUDIODRIVER;

void setup() {
  // Initialize serial communication for debugging
  Serial.begin(115200);
  delay(2000);  // Allow time for the serial monitor to initialize
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);
  LOGLEVEL_AUDIODRIVER = 0;
  Serial.println("Setup starting...");

  Serial.println("Configuring I2C pins...");
  my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES7210ADDR, I2CSPEED,
                 Wire);

  Serial.println("Configuring I2S pins...");
  my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, -1, SDOUT1, 0);

  Serial.println("Initializing audio board...");
  CodecConfig cfg;
  cfg.input_device = ADC_INPUT_ALL;     // Use all available ADC channels
  cfg.output_device = DAC_OUTPUT_NONE;  // No DAC output
  cfg.i2s.fmt = I2S_TDM;                // TDM mode
  cfg.i2s.mode = MODE_MASTER;           // Master mode
  cfg.i2s.bits = BIT_LENGTH_16BITS;     // 16-bit audio
  cfg.i2s.rate = RATE_44K;              // 44.1 kHz sampling rate
  cfg.i2s.channels = CHANNELS2;
  if (!audio_board.begin(cfg)) {
    Serial.println("Audio board initialization failed!");
    // while (1) {} // Stop execution if initialization fails
  }
}

void loop() {
}
`
Setup starting...
Configuring I2C pins...
Configuring I2S pins...
Initializing audio board...
Debug:   AudioBoard::begin
Debug:   DriverPins::begin
Debug:   DriverPins::setupPinMode
Debug:   PinsI2C::begin: 64
Debug:   PinsI2C::begin for 8
Info:    setting up I2C scl: 48, sda: 47
[... 73 lines cut ...]
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x46
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x46 value=0x0
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4B value=0xFF
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4C value=0xFF
Info:    Enable ES7210_INPUT_MIC1
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x14
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4B value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43 value=0x1A
Info:    Enable ES7210_INPUT_MIC2
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x14
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4B value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x44
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x44 value=0x1A
Error:   es7210_start failed
Error:   AudioDriver controlState failed
Error:   setConfig has failed
Error:   AudioBoard::driver::begin failed
Audio board initialization failed!

I have confirmed that there is a valid ES7210 address (7bit 0x40) on the bus. I think it might be the configure issue.
I also notice that under "DribverCommon.h", channels_t only define CHANNELS2, CHANNELS8, CHANNELS16 but no CHANNELS4. I wonder if it is possible to achieve my idea by using this repo?
Any help would be appriciated.

pschatzmann, 2024-12-12:
What processor are you using ?

First try to test the basic functionality with 2 channels and I2S with the ESP32 as master and the ES7210 as slave: configure the AudioBoard via the I2SCodecStream: 

   auto cfg = i2s_in_stream.defaultConfig();
   i2s_in_stream.begin(cfg);

Your scenario has not been tested, so you might need to prepared to resolve a couple of issues and extend the functionality a bit. I suggest that you clone this project, so that you can submit your changes in a pull request.

- Support for CHANNELS4
- Extend the driver to configure the microphones dependent on the channels. 
- You might need to extend the driver to support TDM: I am not sure if this has been implemented

xiqwertyui, 2024-12-13:
```
#include "AudioTools.h"  // install https://github.com/pschatzmann/arduino-audio-tools
#include "AudioTools/AudioLibs/I2SCodecStream.h"
#include "HAL.h"
#include "esp32s3/rom/rom_layout.h"
// I2C configuration
#define SDAPIN CONFIG_SDA  // I2C Data pin
#define SCLPIN CONFIG_SCL  // I2C Clock pin
#define I2CSPEED 100000    // I2C Clock Rate
#define ES7210ADDR 0x40    // Address of ES7210 I2C port

// I2S configuration
#define MCLKPIN CONFIG_ES7210_MCLK   // Master Clock
#define BCLKPIN CONFIG_ES7210_SCLK   // Bit Clock
#define WSPIN CONFIG_ES7210_LRCK     // Word Select (LRCK)
#define SDOUT1 CONFIG_ES7210_SDOUT1  // Data Out 1
#define SDOUT2 CONFIG_ES7210_SDOUT2  // Data Out 2

AudioInfo audio_info(44200, 4, 16);
DriverPins my_pins;
AudioBoard audio_board(AudioDriverES7210, my_pins);
I2SCodecStream i2s_in_stream(audio_board);

extern int LOGLEVEL_AUDIODRIVER;

void setup() {
  // Initialize serial communication for debugging
  Serial.begin(115200);
  delay(2000);  // Allow time for the serial monitor to initialize
  AudioLogger::instance().begin(Serial, AudioLogger::Debug);
  LOGLEVEL_AUDIODRIVER = 0;
  Serial.println("Setup starting...");

  Serial.println("Configuring I2C pins...");
  my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES7210ADDR, I2CSPEED,
                 Wire);

  Serial.println("Configuring I2S pins...");
  my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, -1, SDOUT1, 0);

  auto cfg = i2s_in_stream.defaultConfig();
  i2s_in_stream.begin(cfg);

  Serial.println("Setup completed successfully.");
 
}

void loop() {
 
}
```
Setup starting...
Configuring I2C pins...
Configuring I2S pins...
Debug:   AudioBoard::begin
Debug:   DriverPins::begin
Debug:   DriverPins::setupPinMode
Debug:   PinsI2C::begin: 64
Debug:   PinsI2C::begin for 8
Info:    setting up I2C scl: 48, sda: 47
[  2164][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=47 scl=48 freq=100000
Info:    Setting i2c clock: 100000
Info:    AudioDriver::begin
Info:    AudioDriver::setConfig
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x0 value=0xFF
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x0 value=0x41
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x1F
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x9 value=0x30
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0xA value=0x30
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x23 value=0x2A
[... 67 lines cut ...]
Info:    Enable ES7210_INPUT_MIC1
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x14
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4B value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43 value=0x1A
Info:    Enable ES7210_INPUT_MIC2
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x14
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4B value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x44
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x44 value=0x1A
Error:   es7210_start failed:20
Error:   AudioDriver controlState failed
Error:   setConfig has failed
Error:   AudioBoard::driver::begin failed
[E] I2SCodecStream.h : 220 - bool audio_tools::I2SCodecStream::begin1()
Setup completed successfully.

This gives same error. I'm using ESP32S3 with a custom board. 
 
I've looked into the driver code.
[image]
es7210_read_reg() return 20 and RESULT_OK is 0. So it failed eventually. 
But the function signature of es7210_read_reg() is suppose to return data but not error_t. No too sure if there is something wrong in here.

pschatzmann, 2024-12-13:
I double checked with the [original version from Espressif](https://github.com/espressif/esp-adf/blob/8dffd07f1d0f88c979dc7762bc9ef33ce98ecf5c/components/audio_hal/driver/es7210/es7210.c#L29) and they still do the same thing - though it does not look right to me.
Did you try to replace rc with regv ?

xiqwertyui, 2024-12-13:
Yeah. I also checked original version and some logic in there confused me.
After I changed ret into regv. It seems working (at least no other error messages)

[image]

Setup starting...
Configuring I2C pins...
Configuring I2S pins...
[D] I2SCodecStream.h : 79 - virtual bool audio_tools::I2SCodecStream::begin(audio_tools::I2SCodecConfig)
[D] I2SCodecStream.h : 216 - bool audio_tools::I2SCodecStream::begin1()
[D] I2SCodecStream.h : 253 - void audio_tools::I2SCodecStream::setupI2SPins()
[D] I2SCodecStream.h : 278 - audio_driver_local::Optional<audio_driver::PinsI2S> audio_tools::I2SCodecStream::getI2SPins()
[D] I2SCodecStream.h : 285 - bool audio_tools::I2SCodecStream::beginCodec(audio_tools::I2SCodecConfig)
[D] I2SCodecStream.h : 301 - input: 1
[D] I2SCodecStream.h : 302 - output: 3
[D] I2SCodecStream.h : 317 - BIT_LENGTH_16BITS
[D] I2SCodecStream.h : 351 - RATE_44K
[D] I2SCodecStream.h : 366 - I2S_NORMAL
Debug:   AudioBoard::begin
Debug:   DriverPins::begin
Debug:   DriverPins::setupPinMode
Debug:   pinMode for 6
Debug:   PinsI2C::begin: 64
Debug:   PinsI2C::begin for 8
Info:    setting up I2C scl: 48, sda: 47
[  2182][I][esp32-hal-i2c.c:75] i2cInit(): Initialising I2C Master: sda=47 scl=48 freq=100000
Info:    Setting i2c clock: 100000
Info:    AudioDriver::begin
Info:    AudioDriver::setConfig
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x0 value=0xFF
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x0 value=0x41
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x1F
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x9 value=0x30
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0xA value=0x30
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x23 value=0x2A
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x22 value=0xA
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x20 value=0xA
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x21 value=0x2A
Info:    ES7210 in Slave mode
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x8
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x8 value=0x10
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x40 value=0x43
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x41 value=0x70
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x42 value=0x70
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x7 value=0x20
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x2 value=0xC1
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x2
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x2 value=0xC1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x7 value=0x20
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4 value=0x1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x5 value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43 value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x44
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x44 value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x45
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x45 value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x46
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x46 value=0x0
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4B value=0xFF
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4C value=0xFF
Info:    Enable ES7210_INPUT_MIC1
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x14
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x4B value=0x0
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x43 value=0x10
Info:    Enable ES7210_INPUT_MIC2
Debug:   i2c_bus_read_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x1 value=0x14
[... 59 lines cut ...]
[D] I2SESP32.h : 63 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfigESP32)
[D] I2SESP32.h : 182 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfigESP32, int, int)
[I] AudioTypes.h : 130 -  sample_rate: 44100 / channels: 2 / bits_per_sample: 16
[I] I2SConfigESP32.h : 80 - rx/tx mode: RX_MODE
[I] I2SConfigESP32.h : 81 - port_no: 0
[I] I2SConfigESP32.h : 82 - is_master: Master
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 16
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: Digital
[I] I2SConfigESP32.h : 88 - i2s_format: I2S_STD_FORMAT
[I] I2SConfigESP32.h : 90 - auto_clear: false
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 101 - pin_mck: 6
[I] I2SConfigESP32.h : 103 - pin_bck: 7
[I] I2SConfigESP32.h : 105 - pin_ws: 15
[I] I2SConfigESP32.h : 107 - pin_data: 16
[D] I2SESP32.h : 214 - i2s_driver_install
[D] I2SESP32.h : 230 - i2s_set_pin
[D] I2SESP32.h : 241 - i2s_zero_dma_buffer
[D] I2SESP32.h : 245 - begin - started
Debug:   setVolume: 100
Debug:   ADC can adjust gain

pschatzmann, 2024-12-13:
I committed [some corrections](https://github.com/pschatzmann/arduino-audio-driver/commit/2ed3d0c23c7e8623ffb3856f0bc489db0fbdd601):
-

pschatzmann, 2025-01-08:
What value did you set it to ?
Was it 0b00000010 ?

xiqwertyui, 2025-01-09:
This is true (i.e. 0x02).

pschatzmann, 2025-01-09:
I added the singal_type field to the audio-driver library and made sure that it is filled from the singal_type in the i2s configuration.

In the example sketch above you can use

    auto cfg = i2s_in_stream.defaultConfig();;;
    cfg.signal_type = TDM;
    i2s_in_stream.begin(cfg);

This will set up register 0x12

closed 2025-01-09 by pschatzmann

xiqwertyui, 2025-01-09:
Great work. I will test it later.
However, on the esp32s3 side, I use the native API to configure the I2S interface to setup TDM (e.g. setup TDM channels), and the default I2S_stream configuration may not be available.

pschatzmann, 2025-01-09:
Then it would be 

  CodecConfig cfg;
  cfg.input_device = ADC_INPUT_ALL;     // Use all available ADC channels
  cfg.output_device = DAC_OUTPUT_NONE;  // No DAC output
  cfg.i2s.fmt = I2S_NORMAL;                //  normal mode
  cfg.signal_type = SIGNAL_TDM;        // TDM
  cfg.i2s.mode = MODE_MASTER;           // Master mode
  cfg.i2s.bits = BIT_LENGTH_16BITS;     // 16-bit audio
  cfg.i2s.rate = RATE_44K;              // 44.1 kHz sampling rate
  cfg.i2s.channels = CHANNELS2;

xiqwertyui, 2025-01-10:
In I2S normal mode, works well. But in TDM mode, I2S stream won't start in rx mode properly for some reason.

```
#include "AudioTools.h"  // install https://github.com/pschatzmann/arduino-audio-tools
#include "AudioTools/AudioLibs/I2SCodecStream.h"
#include "HAL.h"
#include "Wire.h"

// I2C configuration
#define SDAPIN CONFIG_SDA  // I2C Data pin
#define SCLPIN CONFIG_SCL  // I2C Clock pin
#define I2CSPEED 400000    // I2C Clock Rate
#define ES7210ADDR 0x40    // Address of ES7210 I2C port

// I2S configuration
#define MCLKPIN CONFIG_ES7210_MCLK   // Master Clock
#define BCLKPIN CONFIG_ES7210_SCLK   // Bit Clock
#define WSPIN CONFIG_ES7210_LRCK     // Word Select (LRCK)
#define SDOUT1 CONFIG_ES7210_SDOUT1  // Data Out 1
#define SDOUT2 CONFIG_ES7210_SDOUT2  // Data Out 2

DriverPins my_pins;
AudioBoard audio_board(AudioDriverES7210, my_pins);
I2SCodecStream i2s_in_stream(audio_board);

extern int LOGLEVEL_AUDIODRIVER;

void setup() {
  // Initialize serial communication for debugging
  Serial.begin(115200);
  delay(2000);  // Allow time for the serial monitor to initialize

  AudioLogger::instance().begin(Serial, AudioLogger::Debug);
  LOGLEVEL_AUDIODRIVER = 0;
  Serial.println("Setup starting...");

  Serial.println("Configuring I2C pins...");
  my_pins.addI2C(PinFunction::CODEC, SCLPIN, SDAPIN, ES7210ADDR, I2CSPEED,
                 Wire);

  Serial.println("Configuring I2S pins...");
  my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, -1, SDOUT1, 0);

  CodecConfig cfg;
  cfg.input_device = ADC_INPUT_ALL;      // Use all available ADC channels
  cfg.output_device = DAC_OUTPUT_NONE;   // No DAC output
  cfg.i2s.fmt = I2S_NORMAL;              // Normal format
  cfg.i2s.singal_type = SIGNAL_DIGITAL;  // TDM
  cfg.i2s.mode = MODE_SLAVE;             // Slave mode
  cfg.i2s.bits = BIT_LENGTH_16BITS;      // 16-bit audio
  cfg.i2s.rate = RATE_44K;               // 44.1 kHz sampling rate
  cfg.i2s.channels = CHANNELS4;
  audio_board.begin(cfg);

  I2SCodecConfig i2s_cfg = i2s_in_stream.defaultConfig(RX_MODE);
  i2s_in_stream.begin(i2s_cfg);
}

void loop() {
  int16_t buf[1024];
  while (1) {
    memset(buf, 0, 1024 * sizeof(int16_t));
    HAL::Mic_Read(buf, NULL, 1024);
    for (int i = 0; i + 4 < 1024; i += 4) {
      printf("%d    ", buf[i]);
      printf("%d    ", buf[i + 1]);
      printf("%d    ", buf[i + 2]);
      printf("%d    \n", buf[i + 3]);
    }
  }
[... 703 lines cut ...]
[I] I2SConfigESP32.h : 83 - sample rate: 44100
[I] I2SConfigESP32.h : 84 - bits per sample: 16
[I] I2SConfigESP32.h : 85 - number of channels: 2
[I] I2SConfigESP32.h : 86 - signal_type: TDM
[I] I2SConfigESP32.h : 90 - auto_clear: false
[I] I2SConfigESP32.h : 92 - use_apll: true
[I] I2SConfigESP32.h : 97 - buffer_count:6
[I] I2SConfigESP32.h : 98 - buffer_size:512
[I] I2SConfigESP32.h : 101 - pin_mck: 6
[I] I2SConfigESP32.h : 103 - pin_bck: 7
[I] I2SConfigESP32.h : 105 - pin_ws: 15
[I] I2SConfigESP32.h : 107 - pin_data: 16
[W] I2SESP32.h : 394 - signal_type undefined
[D] I2SESP32.h : 214 - i2s_driver_install
[D] I2SESP32.h : 235 - Using built in DAC
[D] I2SESP32.h : 241 - i2s_zero_dma_buffer
[D] I2SESP32.h : 245 - begin - started
Debug:   setVolume: 100
Debug:   ADC can adjust gain
E (2219) I2S: i2s_read(2265): RX mode is not enabled
[  2386][E][HAL_MIC.cpp:112] Mic_Read(): Error read i2s0
0    0    0    0    
0    0    0    0    
0    0    0    0    
```

pschatzmann, 2025-01-10:
Be careful: if you use a I2SCodecStream the AudioBoard is automatically set up with it's definitions!

xiqwertyui, 2025-01-10:
I've noticed this. So after begin `I2SCodecStream`, `audio_board.setConfig() ` is used to reconfigure the codec chips.
I just try using `I2SCodecStream` to setup I2S port at ESP32 end.
However, I2SCodecStream does not support TDM now (as mentioned above).
But in native struct `i2s_config_t`, TDM mode is setup through member `.channel_format = I2S_CHANNEL_FMT_MULTIPLE`  instead of `.mode`.
In file `I2SESP32.h`, I notice that there is a toMode() function is trying to setup TDM (maybe in future, because `TDM` is also in `I2SSignalType`), which could be problematic.
[image]
[image]
In current version, if I use below code to start I2S, it just start in default mode (i.e. master, TX).

Besides that, the driver configures the codec chip well.

pschatzmann, 2025-01-10:
I don't quite understand: what ESP32 core version are you using ? 
TDM is only available starting from 3.0 which would be using the I2SESP32V1.h !

If you want to read data you would use:

auto cfg = i2s_in_stream.defaultConfig(RX_MODE);
cfg.signal_type = TDM;
i2s_in_stream.begin(cfg);

The signal_type TDM will make sure that the right driver (DriverPDM) is used-

xiqwertyui, 2025-01-10:
I'm using ESP32S3 with ESP_IDF_VERSION v4.4.6-dirty.
However, I2SESP32V1.h needs `ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(5, 0, 0)`.
[image]
I may not be able to test this unless I upgrade to a newer version of IDF, but as long as the codec is configured well, the rest of the issues don't bother me at all.

Again, very appreciate your help.