pschatzmann/arduino-audio-driver #49: Error compiling v0.1.2

Marco1971Repo, 2025-03-18:
### Problem Description

Everything work as expected using v0.1.1.
Linking v0.1.2 this piece of code:

```
    AudioKitEs8388V1.begin(cfg);    
    auto spi_opt = AudioKitEs8388V1.getPins().getSPIPins(PinFunction::SD);
    if (spi_opt)
    {
        auto spi = spi_opt.value();
        
        if (!SD.begin(spi.cs, *(spi.p_spi)))
            logSuSeriale(F("SD.Begin failed...%d\n"), spi.cs);
    }
```
raise an error during compilation:

src/ESP32_WebRadio.cpp: In function 'void setup()':
src/ESP32_WebRadio.cpp:191:31: error: 'audio_driver::spi_bus_handle_t' {aka 'void*'} is not a pointer-to-object type
  191 |         if (!SD.begin(spi.cs, *(spi.p_spi)))

'audio_driver::spi_bus_handle_t' {aka 'void*'} is not a pointer-to-object type
expression must be a pointer to a complete object type

### Device Description

Esp32 A1S

### Sketch

```cpp
AudioKitEs8388V1.begin(cfg);    
    auto spi_opt = AudioKitEs8388V1.getPins().getSPIPins(PinFunction::SD);
    if (spi_opt)
    {
        auto spi = spi_opt.value();
        
        if (!SD.begin(spi.cs, *(spi.p_spi)))
            logSuSeriale(F("SD.Begin failed...%d\n"), spi.cs);
    }
```

### What is your development environment (with Arduino Core Version information)

Platformio -> Arduino core 3.1.1 based on IDF 5.3.2.241224

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-03-18:
spi_bus_handle_t is a void pointer, you need to cast it to the object your environement supports!
This has been changed to be platform independent from Arduino!

closed 2025-03-18 by pschatzmann

Marco1971Repo, 2025-03-31:
Hello, probably my fault but the error is still there....
Could you clarify or if i have to change my code?
My understaing was your commit was to solve the issue.

pschatzmann, 2025-04-01:
I had to undo the last correction: use a cast  

            SPIClass* spi = (SPIClass*) spi_opt.value();