pschatzmann/arduino-audio-driver #53: Pin or Wire not working on ESP32-S3 on AudioDriverWM8960

derlucas, 2025-04-17:
### Problem Description

I'm trying to setup the AudioDriverWM8960 with a ESP32-S3 Board and i'm getting 

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
`
in i2c_bus_write_bytes

I traced it down to being "getI2C" return nullptr

### Device Description

i tried Waveshare  ESP32-S3 Pico and Wemos Lolin S3 Pro boards.

S3 Pico SCL = 7, SDA = 6
Lolin SCL = 10, SDA = 9

### Sketch

```cpp
#include "AudioBoard.h"
 
DriverPins my_pins;
AudioBoard board(AudioDriverWM8960, my_pins);

void setup() {

  // tried this, without luck
  Wire.setPins(6,7); Wire.begin();

  Serial.begin(115200);

  // Logging does not seem to work either
  AudioDriverLogger.begin(Serial,AudioDriverLogLevel::Debug); 

  delay(1000);
  
  // add i2c codec pins: scl, sda, port

  // does not matter if this:
  my_pins.addI2C(PinFunction::CODEC, 7, 6);

  // or that
  //my_pins.addI2C(PinFunction::CODEC, 7, 6, -1, 100000, &Wire, true);

  // add i2s pins: mclk, bclk, ws, data_out, data_in
  //my_pins.addI2S(PinFunction::CODEC, 0, 16, 15, 18, 17);

  // i tried this, but does not seem to make a difference
  //my_pins.begin();

  // configure codec
  CodecConfig cfg;
  cfg.input_device = ADC_INPUT_LINE1;   // your examples are outdated, too
  cfg.output_device = DAC_OUTPUT_ALL;
  cfg.i2s.bits = BIT_LENGTH_16BITS;
  cfg.i2s.rate = RATE_44K;
  //cfg.i2s.fmt = I2S_NORMAL;
  //cfg.i2s.mode = MODE_SLAVE; 
  board.begin(cfg);   
  
}

void loop() {
  delay(300);
}
```

### Other Steps to Reproduce

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4200a0d9  PS      : 0x00060430  A0      : 0x82008fe1  A1      : 0x3fcebd00  
A2      : 0x3fc96b30  A3      : 0x0000001a  A4      : 0x3fcebd5e  A5      : 0x00000002  
A6      : 0x00000000  A7      : 0x3fc96960  A8      : 0x8200a0d9  A9      : 0x3fcebce0  
A10     : 0x000003e8  A11     : 0x00000014  A12     : 0xffffffff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x4200a0d6:0x3fcebd00 0x42008fde:0x3fcebd30 0x42009017:0x3fcebd50 0x420090a3:0x3fcebd80 0x420094bf:0x3fcebe00 0x42004d86:0x3fcebe20 0x42004c3f:0x3fcebe40 0x42005289:0x3fcebe90 0x4200e88a:0x3fcebec0

### What is your development environment (with Arduino Core Version information)

Platformio 
platform: espressif32
board: waveshare-esp32-s3-pico
framework: arduino

PLATFORM: Espressif 32 (6.10.0) > Waveshare ESP32-S3-Pico (16 MB QD, 2MB PSRAM) HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash 

PACKAGES:
 - framework-arduinoespressif32 @ 3.20017.241212+sha.dcc1105b
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-mkfatfs @ 2.0.1 
 - tool-mklittlefs @ 1.203.210628 (2.3) 
 - tool-mkspiffs @ 2.230.0 (2.30) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5 

Dependency Graph 
-- audio-tools @ 1.1.0+sha.036645b
-- audio-driver @ 0.1.2+sha.993248c

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

pschatzmann, 2025-04-17:
Please provide the decoded backtrace!

derlucas, 2025-04-17:
Hey here it is:

> 0x4200a0d6: i2c_bus_write_bytes at API_I2C_Arduino.cpp:54
> 0x42008fde: i2c_write at mtb_wm8960.cpp:640
> 0x42009017: mtb_wm8960_write_ex(mtb_wm8960_reg_t, unsigned short) at mtb_wm8960.cpp:609
> 0x420090a3: mtb_wm8960_write at mtb_wm8960.cpp:655
> 0x420094bf: mtb_wm8960_init at mtb_wm8960.cpp:403
> 0x42004d86: audio_driver::AudioDriverWM8960Class::begin(audio_driver::CodecConfig, audio_driver::DriverPins&) at Driver.h:956
> 0x42004c3f: audio_driver::AudioBoard::begin() at AudioBoard.h:43
> 0x42005289: audio_driver::AudioBoard::begin(audio_driver::CodecConfig) at AudioBoard.h:59
>  (inlined by) setup() at display.cpp:39
> 0x4200e88a: loopTask(void*) at main.cpp:42

pschatzmann, 2025-04-17:
This is strange: can you double check if you have an actual version: i2c_bus_write_bytes at API_I2C_Arduino.cpp line 54 is a AD_LOGD() statement. This should not result in a backtrace!

derlucas, 2025-04-17:
ah i'm sorry, i modified the lib with many Serial.println() because i could not figure out the logging of your lib.

I reset the code and ran again:

> 0x42009c4c: i2c_bus_write_bytes at API_I2C_Arduino.cpp:53
> 0x42008c6a: i2c_write at mtb_wm8960.cpp:634
> 0x42008c91: mtb_wm8960_write_ex(mtb_wm8960_reg_t, unsigned short) at mtb_wm8960.cpp:604
> 0x42008cf4: mtb_wm8960_write at mtb_wm8960.cpp:647
> 0x42008d26: mtb_wm8960_init at mtb_wm8960.cpp:400
> 0x42003ee3: audio_driver::AudioDriverWM8960Class::begin(audio_driver::CodecConfig, audio_driver::DriverPins&) at Driver.h:941
> 0x42003f66: audio_driver::AudioBoard::begin() at AudioBoard.h:32
> 0x420043d1: audio_driver::AudioBoard::begin(audio_driver::CodecConfig) at AudioBoard.h:44
>  (inlined by) setup() at display.cpp:43
> 0x4200e3ae: loopTask(void*) at main.cpp:42

pschatzmann, 2025-04-17:
I committed a correction: can you double check ?

derlucas, 2025-04-17:
Thank you for the quick response! Your commit made it work.

Thanks a lot!

closed 2025-04-17 by pschatzmann