pschatzmann/arduino-audio-driver #63: Error using I2C

Keniis0712, 2025-08-15:
### Problem Description

I got a repeated log when init:
```
Debug:   i2c_bus_write_bytes: addr=0x40 reglen=1 datalen=1 - reg=0x0 
E (3867) i2c.master: I2C hardware NACK detected
E (3868) i2c.master: I2C transaction unexpected nack detected
E (3872) i2c.master: s_i2c_synchronous_transaction(945): I2C transaction failed
E (3880) i2c.master: i2c_master_multi_buffer_transmit(1207): I2C transaction failed
[  2293][E][esp32-hal-i2c-ng.c:272] i2cWrite(): i2c_master_transmit failed: [259] ESP_ERR_INVALID_STATE
Error:   ->p_wire->endTransmission: 4
```
I have checked my pins, and there's no error. I can communicate with another chip properly
Do I use something wrong?

### Device Description

ESP32-WROVER-32

### Sketch

```cpp
// I2C
#define ES7210ADDR        0x41 // Address of ES7210 I2C port

// I2S, your configuration for the ES7210 board
#define MCLKPIN             32 // Master Clock
#define BCLKPIN             12 // Bit Clock
#define WSPIN               33 // Word select
#define DOPIN               25 // This is connected to DI on ES7210 (MISO)
#define DIPIN               -1 // This is connected to DO on ES7210 (MOSI)

AudioInfo                     audio_info(44200, 2, 16);                // sampling rate, # channels, bit depth
SineWaveGenerator<int16_t>    sine_wave(32000);                        // amplitude
GeneratedSoundStream<int16_t> sound_stream(sine_wave);                 // sound generator
DriverPins                    my_pins;                                 // board pins
AudioBoard                    audio_board(AudioDriverES7210, my_pins); // audio board
I2SCodecStream                i2s_out_stream(audio_board);             // i2s coded
StreamCopy                    copier(i2s_out_stream, sound_stream);    // stream copy sound generator to i2s codec

extern "C" void app_main()
{
  initArduino();

  Serial.begin(115200);
  init_sd();

  vTaskDelay(pdMS_TO_TICKS(1000));

  AudioLogger::instance().begin(Serial, AudioLogger::Debug);
  AudioDriverLogger.begin(Serial,AudioDriverLogLevel::Debug);

  my_pins.addI2C(PinFunction::CODEC, 22, 21, ES7210ADDR, 100000, Wire);
  my_pins.addI2S(PinFunction::CODEC, MCLKPIN, BCLKPIN, WSPIN, DOPIN, DIPIN);

  auto i2s_config = i2s_out_stream.defaultConfig();
  // i2s_config.sd_active = false;
  i2s_config.copyFrom(audio_info);
  Serial.println("Starting I2S output stream...");
  if(!i2s_out_stream.begin(i2s_config)) {
    Serial.println("Failed to initialize I2S output stream");
    while(1);
  }
  Serial.println("I2S output stream initialized");

  if(!sine_wave.begin(audio_info, N_B4)) {
    Serial.println("Failed to initialize sine wave generator");
    while(1);
  }

  while (1)
  {
    copier.copy();
    vTaskDelay(1);
  }
}
```

### What is your development environment (with Arduino Core Version information)

IDF 5.4.2 with arduino-esp32

### I have checked existing issues, discussions and online documentation

- [x] I confirm I have checked existing issues, discussions and online documentation

Keniis0712, 2025-08-15:
What's more, I didn't paste all of my code, but I think these code is enough to show the main problem.

pschatzmann, 2025-08-15:
Either your pins are wrong or you are using a wrong address!

Check with the Arduino I2C Scanner Sketch before using this library....

closed 2025-08-15 by pschatzmann

Keniis0712, 2025-08-15:
I have checked my pins, and I can believe it’s correct by the i2c scanner. As you can see in the latest log, some of the communication success, but others get a NACK. I don’t know why this happened.

---- Replied Message ----
| From | Phil ***@***.***> |
| Date | 08/15/2025 21:59 |
| To | pschatzmann/arduino-audio-driver ***@***.***> |
| Cc | Ken ***@***.***>,
Author ***@***.***> |
| Subject | Re: [pschatzmann/arduino-audio-driver] Error using I2C (Issue #63) |
pschatzmann left a comment (pschatzmann/arduino-audio-driver#63)

Either your pins are wrong or you are using a wrong address!

Check with the Arduino I2C Scanner Sketch before using this library....

pschatzmann, 2025-08-15:
I don't see any success!
What is the output of the scanner ?

pschatzmann, 2025-08-15:
Did you try if you can reproduce the same issue using Arduino ?

Keniis0712, 2025-08-16:
Emm, I think I know why this happend. I should use ES8311 instead of ES7210, and that change fixs all the i2c errors.
But now, I have some new problems. If I add ```my_pins.addPin(PinFunction::MCLK_SOURCE, FROM_SCLK_PIN, PinLogic::Output);```
the log will stuck:
```
[D] I2SCodecStream.h : 79 - virtual bool audio_tools::I2SCodecStream::begin(audio_tools::I2SCodecConfig)
[D] I2SCodecStream.h : 216 - bool audio_tools::I2SCodecStream::begin1()
[D] I2SCodecStream.h : 253 - void audio_tools::I2SCodecStream::setupI2SPins()
[D] I2SCodecStream.h : 278 - audio_driver_local::Optional<audio_driver::PinsI2S> audio_tools::I2SCodecStream::getI2SPins()
[D] I2SCodecStream.h : 285 - bool audio_tools::I2SCodecStream::beginCodec(audio_tools::I2SCodecConfig)
[D] I2SCodecStream.h : 301 - input: 1
[D] I2SCodecStream.h : 302 - output: 3
[D] I2SCodecStream.h : 318 - BIT_LENGTH_16BITS
[D] I2SCodecStream.h : 356 - RATE_48K
[D] I2SCodecStream.h : 367 - I2S_NORMAL
Debug:   AudioBoard::begin
Info:    AudioDriver::begin
Info:    setupI2CAddress: 0x18
Info:    ==> Updating address: 0x18
Info: 
```
I don't know why it stucks in a weird place, and the watchdog doesn't print a stack backtrace.